symbench/electric-circuits #59: Analytics/Assistant Integration

brollb, 2021-02-17:
It would be great to be able to integrate a trained model to assist with model building. There are many different possibilities so I thought I would create this issue to track some of the ideas as well as any discussion about integration.

A couple ideas for analytics integration:
- suggest the next component
- detect errors (or possible anomalies which may be errors)
- search existing models using similarity/dissimilarity

umesh-timalsina, 2021-02-17:
There are couple of issues which are blocking this currently. 
1. #44 is almost complete at its first iteration
2. #52 Should be completed IMO for this integration to happen
3. #23 Is complete with minor quirks, which need to be complete.

brollb, 2021-02-17:
I agree and disagree :) It would be great to have all those things as they are all important but I don't think they necessarily need to block this integration. The way I see it, there are a couple main steps:
- Convert the GME model to a standard format (netlist)
- Pass the netlist to the analytic
- Return the results to the user

At least the first two are only the responsibility of the plugin; the last point is not obvious how exactly the results should be shown. Ideally, this would be a rich but decoupled integration with the visualizer (or perhaps a new visualizer/button that simply showed the results and wasn't coupled to the central visualizer). In the short term, this could be as simple as just adding the predicted next node to the graph or returning a file that contains the distribution over predicted next nodes. Regardless, I agree that #23 is definitely blocking for this issue.

I think it would be good to create a simple example "analytic" to find any possible surprises and such.