symbench/electric-circuits #71: Refactor RecommendNextComponentsMock to support multiple algorithms/models

brollb, 2021-03-04:
Rather than creating a new plugin for each algorithm for recommending the next component, it would be nice to allow users to select the model to use. This would also make it easier to add new models as they won't require creating a whole new plugin.

umesh-timalsina, 2021-03-04:
Makes sense, are you thinking of doing it using the plugin configuration? Such that it can initialize some class based on the configuration?

brollb, 2021-03-04:
I was actually thinking more like this: https://github.com/symbench/electric-circuits/pull/72/files#diff-a20ecfeb8b3ae05b8bea7bb7bc45b5ef18e6e077b80370509ec99aa3946fda69R26-R40

Basically, so you can just add a python file with an __init__.py file that has an `analyze` function in a directory inside `models/`. Then you can just add the name to the metadata as a valid option.

commit 84794149 (2021-03-04): Refactor recommendation plugin to support multiple algs/models. Closes #71 (#72)

* Refactor example analytic to own file

* Rename RecommendNextComponents{Mock,}

* Remove .only

* fix linting issue

* Fix more linting issues

* Fix more linting errors
closed 2021-03-04 by brollb